Watchdog0x/jabridge #1: Cursor does not move?

mattelacchiato, 2025-01-08:
Thanks for this application!

Sadly, I can't move the selection/cursor up/down with the arrow keys.

OS: Arch/Manjaro
Shell: Bash

Watchdog0x, 2025-01-08:
Thank you for your feedback! 

I'm glad you're enjoying the application.

Regarding the issue you reported about the arrow keys not working to move the selection/cursor:

1. I am currently using Debian and haven’t tested the application on Arch/Manjaro.
2. Could you please let me know which terminal emulator you're using? 

This information will help me troubleshoot the issue more effectively.

I'll do my best to investigate and address this as soon as possible.

Thank you for your patience!

mattelacchiato, 2025-01-08:
Oh, I'm sorry! Obviously, I didn't read the README...

Navigation with arrow keys would be more intuitive for me, but I leave this up to you :-)

closed 2025-01-08 by mattelacchiato