Watchdog0x/jabridge #2: please prove build instructions and/or dependency documentation

freggeldoe, 2025-01-14:
project does not build on its own, apparently some `jabra`  library is missing:

```
/tmp/jLink-0.1.1 $ LANG=C go build
# github.com/Watchdog0x/jLink
/usr/lib/go/pkg/tool/linux_amd64/link: running gcc failed: exit status 1
/usr/bin/gcc -m64 -o $WORK/b001/exe/a.out -Wl,--export-dynamic-symbol=_cgo_panic -Wl,--export-dynamic-symbol=_cgo_topofstack -Wl,--export-dynamic-symbol=crosscall2 -Wl,--export-dynamic-symbol=deviceAttachedFunc -Wl,--export-dynamic-symbol=deviceRemovedFunc -Wl,--compress-debug-sections=zlib /tmp/go-link-2851608974/go.o /tmp/go-link-2851608974/000000.o /tmp/go-link-2851608974/000001.o /tmp/go-link-2851608974/000002.o /tmp/go-link-2851608974/000003.o /tmp/go-link-2851608974/000004.o /tmp/go-link-2851608974/000005.o /tmp/go-link-2851608974/000006.o /tmp/go-link-2851608974/000007.o /tmp/go-link-2851608974/000008.o /tmp/go-link-2851608974/000009.o /tmp/go-link-2851608974/000010.o /tmp/go-link-2851608974/000011.o /tmp/go-link-2851608974/000012.o /tmp/go-link-2851608974/000013.o /tmp/go-link-2851608974/000014.o /tmp/go-link-2851608974/000015.o /tmp/go-link-2851608974/000016.o -O2 -g -L/tmp/jLink-0.1.1/lib -ljabra -L/tmp/jLink-0.1.1/lib -ljabra -O2 -g -lpthread -no-pie
/usr/bin/ld: cannot find -ljabra: No such file or directory
/usr/bin/ld: cannot find -ljabra: No such file or directory
collect2: error: ld returned 1 exit status
```

freggeldoe, 2025-01-15:
Thank you, I got the manual setup working (in a way).
I'm building the project as I wanted to take a look at it and I'm not a big fan of executing random binaries downloaded from the internet and as `root` user at that.

Watchdog0x, 2025-01-15:
I completely understand your hesitance about running random binaries from the internet, especially as the root user. It's always a good practice to be cautious about security risks.

It's great that you got the manual setup working!

The root access is only needed for installing dependencies and setting up `udev` rules for Jabra devices. If you check the `install.sh` script, you can also see exactly what it's doing and where to find the shared library.

I’ll close the issue for now, but when I have more time, I’ll add a manual build guide.

I hope you’ll consider giving a star to this GitHub project!

closed 2025-01-15 by Watchdog0x