Watchdog0x/jabridge #3: Speak 510 not found

am4c130d, 2025-03-04:
jlink doesn't detect Speak 510 USB speakerphone

Installed jlink on Debian Testing system with option run without root permissions. 
Plugged in Jabra Speak 510 via USB. 
Launched jlink.
Animation next to Looking For Dongle shows rotating rectangle and only menu option is "Exit" 

Tried with 510 plugged in before jlink launched as well as after jlink launched, same result.

Output from lsusb gives:

ID 0b0e:0422 GN Netcom Jabra SPEAK 510 USB

Watchdog0x, 2025-03-04:
The issue occurs because jlink was built specifically for headsets and does not support the Jabra Speak 510 speakerphone. 
I could add support for it, but since I don't own the device, I wouldn't be able to test it properly. Sorry!

Have you considered looking for alternative tools that might support the Speak 510 on Linux?

Watchdog0x, 2025-03-04:
Hmm, maybe I'll try updating jlink one of these days. I haven't had time to add headset configuration yet, but once that's in, you might be able to configure the Speak 510 as well, if that's what you're looking for ?

am4c130d, 2025-03-04:
jlink was my alternative tool ;-). I'm not aware of any other way to manage the Speak 510 on Linux.

I'd be happy to test if/when you're ready. I was trying to verify I had latest firmware, but config (I don't think that there is much that is configurable) would be good as well. Thank you!

Watchdog0x, 2025-03-04:
Haha, okay! When I get around to updating it, I'll ping the issue for testing. 😊

Watchdog0x, 2025-03-24:
Hey, I'll start updating it tomorrow. Sorry for the wait! I've been busy.

I only have one day to work on it, so I’ll try to finish before I leave for vacation (no WiFi there).

zetneteork, 2026-09-03:
Root cause: the Speak 510 is a USB-direct device (no dongle). jLink currently assumes a Jabra dongle must be present -- the UI spins on "Looking For Dongle" indefinitely for devices that connect directly via USB.

We are implementing a first-scan detection callback (`FirstScanForDevicesDoneFunc`) that will show "No dongle detected" and list any USB-direct devices found instead of hanging.

Tracking at: https://github.com/opentreecz/jLink/tree/integration/upstream

Watchdog0x, 2026-09-03:
RC5 replaces the old dongle-only UI with direct-USB and through-dongle discovery, automatic device access setup, and a service-owned TUI:

https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.5

Please run:

```bash
./jabridge setup
./jabridge status
./jabridge
```

Run it as your normal user and post the model plus VID:PID, but no serial number or Bluetooth address. I am keeping this issue open until this exact model is tested.

Watchdog0x, 2026-09-03:
RC7 supersedes RC5 and RC6 for testing:

https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.7

RC7 keeps the signed executable archive compatible with older self-updaters and publishes the full IPC guide as the separate `IPC.md` release asset. RC5-to-RC7 self-update now passes end to end and also refreshes/restarts the installed service.

Please use RC7 for the requested real-headset read-only test. The hardware checklist and safety boundary are otherwise unchanged.

am4c130d, 2026-09-04:
```
jabridge status
Jabridge 1.0.0-rc.7
1 supported device(s):

Headset: Jabra SPEAK 510 USB
  ID:         0b0e:0422
  Connection: USB
  Firmware:   unavailable (read firmware: address 8: hidraw read timeout; address 3: hidraw read timeout)
```

Some context, I installed it within a distrobox, and ran `jabridge setup` outside the distrobox, because it needed root access for udev adm.

Both inside and outside the distrobox, running `jabridge status` shows the above, running `jabridge` opens the UI, but nothing is shown connected. When I unplug the 510, it briefly appeared as connected, but then the display returns to nothing connected. Reconnecting the 510 doesn't have the brief connected message.

All of these issues may be due to the use of a distrobox. I can retest on another system in a few days, running the executable as a native install.

Appreciate you continuing to make this work - thank you.

Watchdog0x, 2026-09-04:
Thank you, @am4c130d. The unplug timing was the key detail. This was a Jabridge service bug, so I would not blame Distrobox yet.

Root cause: RC7 waited for optional variant and firmware protocol replies before publishing a newly found USB device to `devices.list`. Speak 510 does not answer the headset/controller addresses, so those reads timed out. Unplugging interrupted the waits, which explains why the TUI showed the device only briefly as it disappeared.

Fixed in RC8:

https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.8

- sysfs identity is published to IPC immediately;
- optional protocol enrichment happens afterward with short timeouts;
- `devices.list` never performs hardware I/O;
- direct-USB devices cannot be removed by dongle-child cleanup.

Please update on the host first:

```bash
jabridge update --prerelease
jabridge service restart
jabridge ipc devices
jabridge
```

`ipc devices` should return the Speak 510 immediately, and the TUI should keep it visible through unplug/reconnect. Running `jabridge setup` outside Distrobox was reasonable because the udev rule belongs to the host.

The installed-firmware read is still a separate open item. The public catalog identifies PID `0b0e:0422` as Speak 510 and reports latest firmware `2.32.8`, but RC8 does not pretend to know the installed version when this model rejects the read-only headset addresses. Please do not try firmware installation. Once detection is confirmed, I will keep this issue open for a narrow read-only Speak 510 firmware/settings probe.

The RC8 race/static/vet/lint checks, signed release verification, RC7-to-RC8 self-update, and live service/TUI regression on Link 380 all pass. We still need your real Speak 510 result to confirm the hardware fix.

Watchdog0x, 2026-09-04:
@am4c130d, RC10 is now the best build for the Speak 510 retest:

https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.10

One correction to my RC8 reply: a `permission denied` error means the udev rule blocked opening hidraw. Your `hidraw read timeout` means the node opened, but that interface/address did not answer. RC10 now reads every matching HID descriptor first and probes only interfaces that declare the Jabra management report, so it will not choose an audio or button interface merely because it has the same USB ID.

Please run natively on the host as your normal user:

```bash
jabridge update --prerelease
jabridge setup
jabridge service restart
jabridge diagnose
jabridge ipc devices
jabridge status
jabridge
```

Expected minimum result: `ipc devices`, `status`, and the TUI keep the Speak 510 visible immediately and remove/restore it on USB unplug/reconnect. `diagnose` may still say that no control reply was found; that would be a separate read-only protocol gap, not a detection failure.

Please post the output without a serial number or Bluetooth address. Do not run `firmware install`, pairing, reset, or setting changes.

Watchdog0x, 2026-09-04:
@am4c130d, RC14 has the fuller debug report for your Speak 510:
https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.14

Please run natively on the host as your normal user:

```bash
jabridge update --prerelease
jabridge setup
jabridge debug --output speak510-host.txt
```

Continue with debug even if setup fails. It records service/IPC failures, USB visibility, HID report fields, native control reads and firmware metadata. In a terminal it also prompts for 20 seconds of button observation; press one control at a time. It records event names and changed bit positions without raw input values or serials.

If useful, run the same command in Distrobox with another filename. The report marks container context without assuming it caused the failure. Please attach the files and say whether the TUI stays connected through USB unplug/reconnect.

Speak 510's public model data identifies firmware protocol 1. Native flashing for that family is not implemented. A firmware timeout or missing management report remains separate from USB detection and permissions. This issue stays open until the actual device results confirm each operation.

am4c130d, 2026-09-05:
from the host:

```
bin/jabridge/jabridge_1.0.0-rc.14_linux_amd64/jabridge update --prerelease
Jabridge 1.0.0-rc.15 is available: https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.15
Downloading and verifying jabridge_1.0.0-rc.15_linux_amd64.tar.gz...
Updated Jabridge to 1.0.0-rc.15
```
```
bin/jabridge/jabridge_1.0.0-rc.14_linux_amd64/jabridge setup
Headset access is ready. Starting your user service...
Jabridge is installed and will start automatically when you sign in.
Device access is ready.
```

https://pastebin.com/VFs1u8JF

from distrobox:

```
./jabridge_1.0.0-rc.14_linux_amd64/jabridge update --prerelease
Jabridge 1.0.0-rc.15 is already up to date.
```
```
./jabridge_1.0.0-rc.14_linux_amd64/jabridge setup
Headset access is ready. Starting your user service...
Jabridge is installed and will start automatically when you sign in.
Device access is ready.
```

https://pastebin.com/V30rXMiD

Hope this is useful.

Watchdog0x, 2026-09-05:
Thank you—this report was useful. It confirms that the Speak 510 is detected immediately by USB and owned by the running service. Distrobox is not the main detection failure.

It also found two separate blockers:

1. `setup` said access was ready because hidraw was accessible, but the current udev rule was missing and `/dev/input/event15` remained permission-denied. That is a Jabridge setup bug.
2. Speak 510 exposes a bidirectional 33-byte vendor transport on HID report 2. Jabridge currently supports the 64-byte report-5 GNP transport, so firmware/settings reads cannot work by reusing the Link/headset framing.

RC16 fixes the setup check and makes the report explain these boundaries directly:
https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.16

Please run this on the host, as your normal user:

```bash
./jabridge update --prerelease
./jabridge setup
```

Accept the one administrator prompt. Reconnect the Speak 510 once if setup asks, then run:

```bash
./jabridge debug --output speak510-rc16-host.txt
```

During the 20-second prompt, press each button once. Please attach the report and say whether the Speak 510 stays visible in the TUI after unplug/reconnect. A Distrobox report is optional; the host report is the important one.

Keep this test read-only. RC16 does not enable Speak 510 firmware installation or settings writes. The report should now show the candidate vendor reports and the six public model variants without guessing which one is your device.

am4c130d, 2026-09-06:
https://pastebin.com/5UTQdaMc

Everything ran error free this time. 

I ran the app - the TUI shows the Speak510, I can go to settings (nothing configurable) and firmware - it shows the latest available, but doesn't know what's loaded on the device - I quit the app before doing anything.

Watchdog0x, 2026-09-07:
Thank you, @am4c130d. Your RC17 report confirms that USB/TUI detection, setup permissions and volume-button events now work.

RC19 includes the native transport fixes and forced permission repair:
https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.19

One correction to my earlier reply: report 2 does not by itself mean a different management protocol. The management usage is `FF00:0001`; its report number and size vary. RC18 now selects that usage, sends correctly sized reports and reassembles split replies. It also fixes three-byte variant parsing and adds the model-filtered call-button action, Bluetooth name and speed-dial settings, including writes with read-back verification.

Please update, force a permission refresh, then collect:

```bash
./jabridge update --prerelease
./jabridge setup --force
./jabridge debug --output speak510-rc19.txt
./jabridge status
./jabridge settings
```

The important checks are the actual device variant, installed firmware and current settings. If editable settings appear and you want to test a change, record the original value, choose a value shown in the list, verify it, then restore the original. Text settings use the CLI; boolean and choice settings can be changed in the TUI.

I verified reads and a reversible setting write on Link 380 before pushing. Speak 510 still needs your physical confirmation, so this issue stays open for native firmware/settings support. Do not try firmware installation: protocol 1 is a separate, unfinished updater path.

am4c130d, 2026-09-07:
```
jabridge status
Jabridge 1.0.0-rc.19
1 detected Jabra device(s):

Headset: Jabra SPEAK 510 USB
  ID:         0b0e:0422
  Connection: USB
  Variant:    08-01
  Battery: 71% (charging)
  Firmware:   2.32.0
```
```
jabridge settings
headset: Jabra SPEAK 510 USB
  headset.call-button = None (editable; choices: None, Speed dial, Cortana)
  headset.voice-prompts = Tones (editable; choices: Tones, Voice)
  headset.bluetooth-name = Jabra SPEAK 510 USB (editable)
  headset.speed-dial =  (editable)
```
Pastebin is offline for maintenance - https://gist.github.com/am4c130d/03be3b11ee4d1fb0a0a7c55f52086442 for the debug report.

The TUI now reports the firmware version (I didn't try to upgrade per your comment, but the TUI reported that a newer version was available), and the menu options all appear to work.

Watchdog0x, 2026-09-07:
@am4c130d, thank you. Jabridge now reads your Speak 510’s firmware, battery and all four settings. The volume buttons and service also work. You do not need to run setup again.

Did you change any settings, or only open the menus?

If voice prompts still show Tones, please try these commands when you are not in a call. Run them one at a time. They change Tones to Voice, then restore Tones.

```bash
jabridge settings set headset.voice-prompts Voice
jabridge settings
jabridge settings set headset.voice-prompts Tones
jabridge settings
```

Please tell us whether both changes say “verified” and actually work. If a command fails, stop and share the error and a new debug report.

Your installed firmware is 2.32.0. Version 2.32.8 is available to download, but Jabridge cannot install it on Speak 510 yet. That part still needs code, so please do not try a firmware update through Jabridge yet.

I have updated the issue with your results. Thank you for helping us test.

am4c130d, 2026-09-07:
I toggled the two settings (call button and voice prompts)  in the TUI and they reported as changed, I didn't actually verify them though.

I issued the set commands, 
```
jabridge settings set headset.voice-prompts Tones
headset.voice-prompts = Tones (verified by service)
```
```
jabridge settings set headset.voice-prompts Voice
headset.voice-prompts = Voice (verified by service)
```

no errors, but the only voice prompt appears to be connect to bluetooth, and that stayed voice irrespective of the setting.

Watchdog0x, 2026-09-07:
Thank you. This confirms that Jabridge writes the value and reads it back. It does not yet confirm that the spoken announcement changes.

I checked the setting values against Jabra’s definition and they match. One useful check is to select Tones, turn the Speak 510 off and on while not in a call, then run `jabridge settings` again. Does it still show Tones, and does the Bluetooth announcement still use a voice?

Please do not factory reset it. I am keeping this behavior open until we understand it. I am also working on the missing Speak 510 firmware updater, but it is not ready for a hardware test yet.

am4c130d, 2026-09-07:
BT stays voice prompt. But the setting appears to stay set after the power cycle:

```
jabridge settings
headset: Jabra SPEAK 510 USB
  headset.call-button = Cortana (editable; choices: None, Speed dial, Cortana)
  headset.voice-prompts = Tones (editable; choices: Tones, Voice)
  headset.bluetooth-name = Jabra SPEAK 510 USB (editable)
  headset.speed-dial =  (editable)
```

Watchdog0x, 2026-09-07:
Thank you. This confirms that the setting is saved and survives a restart.

I found a useful note in [Jabra’s manual, page 13](https://www.jabra.com/_/media/Jabra_VXi_Product-Documentation/Jabra-SPEAK-510-Series/User-Manuals/RevM/Jabra-Speak-510-user-manual_EN_English_RevM.pdf#page=13): some voice prompts remain enabled even when voice guidance is off.

The Bluetooth announcement may be one of those messages. The manual does not name every exception, so I cannot confirm that exact message yet. But hearing it does not by itself show that the setting failed.

I am recording the setting write, readback and persistence as passed. No more setup, reset or repeated power cycling is needed for this check.

am4c130d, 2026-09-07:
Thank you for continuing to drive this forward - much appreciated.

Watchdog0x, 2026-09-07:
@am4c130d, RC22 is ready: https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.22

It shows Voice guidance as On or Off. Off means Tones and On means Voice. The old commands still work. The help text now explains that some voice prompts may remain. Your tests confirm that the setting is written, read back and saved after restart.

Debug now records the device reply and matching readback separately. It does not claim that every button or spoken message was tested.

The missing native firmware path is now implemented as a preview shared with Speak 410, 710 and 810. It passed file checks and simulated transfers, but I do not have a Speak 510 to test the actual update here. If you want to try the first hardware update, connect it directly by USB, close other Jabra tools and do not start during a call:

```bash
jabridge update --prerelease
jabridge setup
jabridge firmware download
jabridge firmware verify ./firmware/Jabra_SPEAK_510_USB-v2.32.8-vector.zip
jabridge firmware install ./firmware/Jabra_SPEAK_510_USB-v2.32.8-vector.zip
```

Use the exact file name printed by download if it differs. Installation asks you to type INSTALL. Leave the device plugged in. If it fails, keep the same file and USB port and share the error before further tests. The [guide](https://github.com/Watchdog0x/jabridge/blob/v1.0.0-rc.22/docs/USB_DFU.md) explains recovery retry.

RC21 was withdrawn because the old updater rejected extra documentation files in its archive. RC22 restores the compatible layout. That failure did not replace the installed binary or change firmware.

Watchdog0x, 2026-09-09:
@am4c130d, thank you for testing. Your reports confirm that Speak 510 is detected through USB, reads its firmware and battery, shows its settings, and saves voice guidance across a restart.

I am closing the original detection and settings problem. This does not mark firmware installation, recovery or every physical button as tested. Those remaining hardware checks are tracked in #36.

Jabridge 1.0.0 is being prepared. It is not published yet.

closed 2026-09-09 by Watchdog0x