Watchdog0x/jabridge #4: jabra evolve2 65 over bluetooth

Matthias199, 2025-03-24:
Hey Jlink does not find it :( its connect over bluetooth

any ideas :)

kevin-messer-jemmic, 2025-05-12:
@Matthias199 and @Watchdog0x I just got a Jabra Evolve 2 65 and it worked out of the box, I did not had to do anything for it to be recognized. I just plugged the dongle and that's it. I am on Ubuntu 25.04.

Atem18, 2025-05-13:
@Matthias199 and @Watchdog0x I just got a Jabra Evolve 2 65 and it worked out of the box, I did not had to do anything for it to be recognized. I just plugged the dongle and that's it. I am on Ubuntu 25.04.

zetneteork, 2026-09-03:
This is related to #3, #5, #20, and #25. jLink currently requires a Jabra dongle -- devices paired directly via system Bluetooth are not managed by the Jabra SDK in the same way.

We are implementing a no-dongle detection mode that shows a clear message when no dongle is found, and lists any USB-direct devices detected. Full system-Bluetooth-direct support would be a larger feature for a future version.

Tracking at: https://github.com/opentreecz/jLink/tree/integration/upstream

Watchdog0x, 2026-09-09:
Direct system-Bluetooth headset control is not implemented in 1.0.0. USB and a headset connected through a Jabra Link dongle are different paths. I am closing this request as deferred, not fixed; the limitation remains in the roadmap #39. PipeWire Bluetooth audio controls do not add native headset settings support.

closed 2026-09-09 by Watchdog0x