Watchdog0x/jabridge #5: Engage 50 II link support?

christf, 2025-03-26:
The above device is not detected with jlink. It would be great to be able to reprogram its buttons and configure the device.

Danfro, 2025-03-28:
I was just going to try the app with the same device when I saw this issue. Thanks, saves me trying in vain. So I will wait for the time being. But yes, would be nice to see it supported.

Watchdog0x, 2025-04-07:
If I get the time, I might add support for it. My focus so far has been only on devices using the Jabra dongle, since that’s how my headset works.

zetneteork, 2026-09-03:
Same root cause as #3 and #25 -- USB-direct devices without a Jabra dongle are not handled by the current UI. We are implementing first-scan detection that will show detected USB devices instead of spinning "Looking For Dongle" forever.

Whether all features work with the Engage 50 II specifically would need hardware testing, but at minimum the device will be detected and shown.

Tracking at: https://github.com/opentreecz/jLink/tree/integration/upstream

Watchdog0x, 2026-09-03:
@Danfro confirmed RC3 detects the Engage 50 II + Link controller as USB `0b0e:4052`. RC5 now adds the one-command permission setup, controller firmware probing, model-filtered headset/controller settings, and supported Link-button choices:

https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.5

I am keeping this issue open until the read-only RC5 settings list is confirmed on the real controller. Please run `./jabridge setup`, then `./jabridge settings`, without changing a value, and report which controls appear. Write testing is not requested on an everyday device.

Watchdog0x, 2026-09-03:
RC7 supersedes RC5 and RC6 for testing:

https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.7

RC7 keeps the signed executable archive compatible with older self-updaters and publishes the full IPC guide as the separate `IPC.md` release asset. RC5-to-RC7 self-update now passes end to end and also refreshes/restarts the installed service.

Please use RC7 for the requested real-headset read-only test. The hardware checklist and safety boundary are otherwise unchanged.

Watchdog0x, 2026-09-04:
Current status: the detailed RC11 test in #36 confirms Engage 50 II detection as `0b0e:4052` with the Link controller and `0b0e:4056` without it. PipeWire sees output/microphone nodes in both cases. Installed firmware/settings remain unavailable and setup times out starting the service.

This issue is the canonical Engage support task. The existing test discussion stays in #36: https://github.com/Watchdog0x/jabridge/issues/36#issuecomment-5545909402

Remaining: collect updated debug reports in both configurations; resolve startup and management-interface failures; confirm sidetone and button/wheel events; implement the missing call integration. No Engage firmware installation is qualified by the Link 380 owner's test.

Watchdog0x, 2026-09-04:
@Danfro, RC14 is ready for the next Engage 50 II report:
https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.14

```bash
./jabridge update --prerelease
./jabridge setup
./jabridge debug --output report.txt
```

The report now includes all exposed setting IDs and choices, native read results, HID fields/activity, service/IPC failure categories, firmware metadata/cached-file checks, and specific next steps. Interactive runs include a short button-observation window; press one control at a time. `--buttons=false` skips it.

No raw input values or private device identities are saved. A successful read is not a successful write/firmware test. Unknown fields and missing native implementations stay explicit.

Please use different report filenames for direct USB, through-Link, and host/container runs. Continue with debug even when setup fails. The new Evolve3 definitions and Speak/Engage paths still require real hardware validation.

Danfro, 2026-09-07:
I am sorry for the delay, but I found a few minutes to test RC19 and grab the screenshots from JabraBridge on Windows.

1. Test RC19

Trying to do the setup did fail with this message:

```
./jabridge setup
Device access is ready. Starting your user service...
jabridge: service did not become ready: context deadline exceeded
LoadState=loaded
ActiveState=activating
SubState=auto-restart
Result=exit-code
ExecMainCode=1
ExecMainStatus=218
Run jabridge debug --output jabridge-debug.txt and share that file
```
[jabridge-debug.txt](https://github.com/user-attachments/files/31926877/jabridge-debug.txt)

When doing the setup with the `--force` parameter, the service did start. 
```
./jabridge setup --force
Refreshing device access with sudo...
[sudo: authenticate] Passwort:                        
Device access is ready. Starting your user service...
jabridge: service did not become ready: context deadline exceeded
LoadState=loaded
ActiveState=activating
SubState=auto-restart
Result=exit-code
ExecMainCode=1
ExecMainStatus=218
Run jabridge debug --output jabridge-debug.txt and share that file
```

Although trying to start `./jabrabridge` does bring this error:
```
./jabridge
Starting Jabridge service...
jabridge: service did not become ready: context deadline exceeded
LoadState=loaded
ActiveState=activating
SubState=auto-restart
Result=exit-code
ExecMainCode=1
ExecMainStatus=218
Run jabridge debug --output jabridge-debug.txt and share that file
```
[jabridge-debug2.txt](https://github.com/user-attachments/files/31927243/jabridge-debug2.txt)

2. Screenshots

It's quite a long list of screenshots, but I hope they do answer your questions.

[screenshots.zip](https://github.com/user-attachments/files/31927080/screenshots.zip)

I hope that helps.

Watchdog0x, 2026-09-07:
@Danfro, thank you for both reports and the screenshots. This was a bug in our service configuration, not missing headset permissions. Sorry for the repeated setup attempts.

Both reports show HID and button access ready. `218/CAPABILITIES` means systemd failed before Jabridge started; the `auto-restart` state was a retry loop, not a working service. Forcing sudo again could not fix that.

[RC20](https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.20) removes the incompatible user-service restrictions and makes debug explain this failure. Please run these as your normal user:

```bash
./jabridge update --prerelease
./jabridge service restart
./jabridge debug --output engage-rc20-with-controller.txt
```

Run the second command even if the old updater reports a service error. It installs the corrected service unit. No `setup --force`, root-run app, or disabled host security is needed when device access already passes.

I reproduced the capability failure locally and tested the corrected service, IPC, Link 380 reads and TUI before publishing. Your Ubuntu host and Engage headset still need confirmation. If startup works, please also try `./jabridge settings` and repeat the report without the controller, with another filename.

The screenshots were useful: the [settings comparison](https://github.com/Watchdog0x/jabridge/blob/codex/native-go-rewrite/docs/ENGAGE_SETTINGS.md) records missing names, second speed dial, IntelliTone level, full mute control and headset-scoped softphone integration. Those are not all implemented yet. Keeping this issue open for the retest.

Watchdog0x, 2026-09-08:
@Danfro, thank you for the screenshots. I used them to check the missing settings. The [new preview](https://github.com/Watchdog0x/jabridge/releases) now lets you pick from a list or edit text in the menu. Some settings still need work.

Please try your Engage 50 II again:

```bash
./jabridge update --prerelease
./jabridge service restart
./jabridge
```

After closing the menu, run:

```bash
./jabridge debug --output engage-with-controller.txt
```

Please attach the report. If you have time, repeat without the controller and use a different filename. Even if the menu fails, please run debug. No forced setup is needed when device access already works.

I tested the service and menu on my Link 380. Your headset still needs this check. No reset or firmware update is needed.

Danfro, 2026-09-08:
@Watchdog0x thanks so much for your quick and instant work on this software. I can't keep up with your speed and sometimes need to go and sleep. 😅 

First test **with** controller:

Only two settings options available.

[image]

[rc23_engage-with-controller.txt](https://github.com/user-attachments/files/31937433/rc23_engage-with-controller.txt)

 Second test **without** the controller:

A long list of settings available. Need to do two screenshots to get them all.

[image]

[image]

I tried changing a few settings. After pressing ENTER I got the on-screen confirmation, that the changes where done. Reopening the software seems to have kept the settings. So looks like all settings are stored on the headset itself, not on the controller. Although I think the two sets of settings should be the other way round with-without controller.

[rc23_engage-without-controller.txt](https://github.com/user-attachments/files/31937518/rc23_engage-without-controller.txt)

Watchdog0x, 2026-09-08:
@Danfro, no rush. Sleep is still a supported feature 😄 Thank you for both tests and the screenshots!

The service is working now. Without the controller, the report confirms that sidetone, sidetone level and music mode changed on the headset and were read back correctly.

With the controller, Jabridge cannot match the device profile and several setting reads fail. That explains why settings are missing. It does not mean the controller only supports two settings.

Reopening the app is a good check, but it does not yet tell us what survives unplugging the headset.

The controller path still needs work. These reports help narrow it down. No need to repeat setup or run more tests for now.

Watchdog0x, 2026-09-08:
@Danfro, [RC24 is ready to test](https://github.com/Watchdog0x/jabridge/releases). It now checks the headset and controller separately and gives them separate settings menus.

Please try both connected first:

```bash
./jabridge update --prerelease
./jabridge
```

Close the menu, then run:

```bash
./jabridge debug --output engage-both.txt
```

When you have time, repeat with only the headset, then only the controller. Save those reports as engage-headset.txt and engage-controller.txt. Please attach the three files and tell us which settings appear.

Some controller settings need the headset connected. Controller-only support still needs this test, so please report what actually happens. No reset, firmware update or setting change is needed.

Automated checks and my Link 380 reads passed. Your Engage hardware is the next check. No rush!

Danfro, 2026-09-08:
Test with RC24

**Controller + headset:**

Headset and controller settings provided.

_Controller settings_
[ON     ] SmartRinger
[MEDIUM ] Controller ringer volume
[TONE 1 ] Controller ringtone
[CALL HANDLING] Call button
[MUTE   ] Mute button
[NONE   ] Three-dot button
[BUSYLIGHT] Four-dot button
Speed dial number:                
Second speed dial number:

_Headset settings_
[ON     ] Sidetone
[OFF    ] In-call busylight
[ON     ] Music mode
[ON     ] Reverse left and right audio
[ON     ] Button sounds
[OFF    ] Ringtone in headset
[ON     ] Softphone integration
[0 DB   ] Sidetone level
[BASIC PEAKSTOP] Audio protection
[85 DB  ] IntelliTone level
[OFF    ] Mute reminder
[NORMAL ] Sound mode

[rc24_engage-with-controller.txt](https://github.com/user-attachments/files/31974041/rc24_engage-with-controller.txt)

**Only controller**

[image]

[rc24_only-controller.txt](https://github.com/user-attachments/files/31974315/rc24_only-controller.txt)

**Only headset**

Headset settings provided as above.

[rc24_only-headset.txt](https://github.com/user-attachments/files/31974366/rc24_only-headset.txt)

Watchdog0x, 2026-09-09:
@Danfro, thank you for the reports and screenshots. RC24 now shows separate headset and Link Call Control settings, with headset-only settings also reported. I am closing this older report as we prepare 1.0.0. This does not claim every physical button or controller-only behavior is verified. Remaining model work is tracked in #39, and firmware/recovery tests in #36. Please open a new issue with a fresh debug report if the new release still has a problem.

closed 2026-09-09 by Watchdog0x