Watchdog0x/jabridge #6: Build instructions

christf, 2025-03-26:
How do I build this code?
I tried go build, but

```
$ go build
# github.com/Watchdog0x/jLink
/usr/lib/go-1.19/pkg/tool/linux_amd64/link: running gcc failed: exit status 1
/usr/bin/ld: cannot find -ljabra: Datei oder Verzeichnis nicht gefunden
/usr/bin/ld: cannot find -ljabra: Datei oder Verzeichnis nicht gefunden
collect2: error: ld returned 1 exit status

$ ls -l /usr/lib/jabra/libjabra.so
-rw-r--r-- 1 root root 1651992 26. Mär 17:57 /usr/lib/jabra/libjabra.so

$ echo $LD_LIBRARY_PATH 
/usr/lib:/usr/lib/jabra:/usr/lib/jabra/libjabra.so

```

Watchdog0x, 2025-04-07:
https://github.com/Watchdog0x/jLink/issues/2

zetneteork, 2026-09-03:
This is now addressed. On the `integration/upstream` branch we have:
- Build instructions in the README ("Building from Source" section)
- A `Makefile` with targets: `make extract-lib`, `make build`, `make lint`, `make help`
- `setup-lib.sh` from PR #27 for setting up `lib/libjabra.so`
- The hex-encoded library in `install.sh` can be extracted with `make extract-lib`

See PR #31: https://github.com/Watchdog0x/jLink/pull/31
Tracking at: https://github.com/opentreecz/jLink/tree/integration/upstream

zetneteork, 2026-09-03:
Build instructions are now available in the README ("Building from Source" section) and via the Makefile (`make extract-lib && make build`). See PR #31 and PR #27 (setup-lib.sh). This can be closed.

Watchdog0x, 2026-09-03:
Resolved in Jabridge v1.0.0-rc.2 and PR #33. The native Go build no longer needs libjabra.so, CGo, or GCC. Run make build with Go 1.23.2 or newer; the result is dist/bin/jabridge. PR #33 is still awaiting real-headset validation before merge, but the source and signed RC2 are available now. Please reopen if the RC2 source build fails.

closed 2026-09-03 by Watchdog0x