Watchdog0x/jabridge #7: What is the origin of the embedded .so file from the install.sh?

christf, 2025-03-26:
Where is this from? What does it contain?

draggeta, 2025-04-03:
Wow, yeah, just came across this project and I cannot say if it is malicious or not, but it does raise suspicion.

Watchdog0x, 2025-04-07:
Hello,

I'm back again.

The .so file comes from the official Jabra SDK, available at: https://developer.jabra.com/sdks-and-tools/linux

It's embedded because Jabra's rules don't allow sharing it on GitHub.

christf, 2025-04-07:
It is being shared on GitHub when it is part of the shell script. Disguising it does not change that. 
Maybe better provide a download link instead like early Nvidia installer did? This should both: scuttle concerns about malicious content as well as follow the distribution rules as per the original author.

Watchdog0x, 2025-04-07:
Thanks for the feedback,  totally fair point.

That said, I don’t have the time or resources to spin up a dedicated server just for hosting the files. I’m doing this in my free time, and out of my own pocket.

Originally, I built this tool just for myself. A friend showed interest, so I decided to share it. I’ll look into better ways to handle that without adding too much overhead.

Appreciate you bringing it up.

christf, 2025-04-07:
> Thanks for the feedback,  totally fair point.
>
> That said, I don’t have the time or resources to spin up a dedicated server just for hosting the files. I’m doing this in my free time, and out of my [...]

I wonder if jabra would embrace making it available for download if they were asked. 
Being able to upgrade firmware and control Jabra devices from Linux should be in their interest especially if someone else is spending the time and energy to build the tools.

christf, 2025-04-07:
> Wow, yeah, just came across this project and I cannot say if it is malicious or not, but it does raise suspicion.

I extracted the so-file from the setup.sh from this repository and calculated a checksum. I also downloaded the so file from the gnaudio-based repo linked in #2 and calculated the checksum. They are the same. I used md5. 

I will share the checksums and the approach in a bit. I do not see a problem (beyond the uncertainty which can be resolved by a few clarifications in prominent spots) here.

christf, 2025-04-15:
what the install.sh left open is the origin of the file. This is now clarified, thank you for you efforts in helping clarify.

closed 2025-04-15 by christf