Watchdog0x/jabridge #9: Gui support

Atem18, 2025-04-29:
Hi, thanks for this software, it works well with my Jabra Evolve 65 linked with my Jabra Link 370.

The CLI works well but I would like to have a small applet in my gnome desktop (can be compatible with KDE as well) to quickly see the battery percentage.

I can have a look on how to do it but would be open to have it directly in the repo or should this be a separate project ?

Thanks a lot in advance !

Watchdog0x, 2025-04-29:
Hi Atem18,

Thanks for the suggestion

I really like the idea. I’ve actually been thinking along the same lines, especially as we get closer to version 1.0.0, though there's still a way to go.

At the moment, it's not possible to build an applet like that because the project doesn’t yet expose a usable API for third-party tools. My long-term goal is to create a proper JabraAPI service managed by systemd, likely using a Unix socket or shared memory (IPC). The CLI would then act as a client to that service, making third-party integrations possible.

Once that’s in place, things like GNOME or KDE desktop applets would definitely be doable.

That said, this is a side project I work on in my free time, which is pretty limited right now. If I had the chance to work on it full-time, I absolutely would.

Still, progress is happening. So far, I’ve added:
1. Settings support
2. Automatic firmware checking
3. Firmware download and update firmware
4. and more...

The next step is updating the UI. Once that’s done, I’ll push out a new release (it’ll still be version 0.2).

If you have more time than I do and want to take this on, feel free to build on the Go-based JabraAPI in the repo. Whether you integrate it here or spin off a separate project, I’d be happy to link to it in the README.

Thanks again for your interest!

Atem18, 2025-04-29:
Hi @Watchdog0x, thanks for the answer. I probably have as much as free time as you do so I will check in the direction that you pointed and see what I can do.

zetneteork, 2026-09-03:
This is a longer-term feature that requires the daemon/IPC architecture described in the TODO list. Not something that can be implemented right now.

However, the infrastructure to support this effort is now in place on the `integration/upstream` branch:
- CI/CD with GitHub Actions (automated builds, releases)
- RPM and DEB packaging
- Contribution templates and guidelines
- Dependabot for dependency updates

When the daemon service is ready, these tools will make it much easier to build and distribute an applet.

Tracking at: https://github.com/opentreecz/jLink/tree/integration/upstream

Watchdog0x, 2026-09-03:
@Atem18 the applet prerequisite is now available in RC5: a persistent per-user service, event subscriptions, reconnecting client, and simple JSON IPC commands.

Start with:

```bash
jabridge ipc devices
jabridge ipc battery
jabridge ipc watch
```

The easy guide is in `docs/IPC.md`. I am keeping this issue open for the actual GNOME/KDE applet; an applet can now be built without opening the hardware itself.

Watchdog0x, 2026-09-03:
RC7 supersedes RC5 and RC6 for testing:

https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.7

RC7 keeps the signed executable archive compatible with older self-updaters and publishes the full IPC guide as the separate `IPC.md` release asset. RC5-to-RC7 self-update now passes end to end and also refreshes/restarts the installed service.

Please use RC7 for the requested real-headset read-only test. The hardware checklist and safety boundary are otherwise unchanged.

Watchdog0x, 2026-09-09:
@Atem18, the shared service and IPC are now available, so a desktop applet no longer needs its own device backend.

For 1.0 we will keep GNOME, KDE, Hyprland and other frontends as separate projects. Jabridge provides the service, terminal menu and CLI. The release includes an IPC guide with a small working example.

I am closing the request for a bundled applet as not planned, not as an applet we have already built. If you make one, please share it so we can link to it. Thank you for the idea!

closed 2026-09-09 by Watchdog0x