Watchdog0x/jabridge #10: Support for Zorin

AMDrewtaylor, 2025-05-05:
Was looking to set this up on my system running Zorin OS, however it isn't currently compatible. 

Would love to be able to use this! 

Evolve2 85 User

Watchdog0x, 2025-05-13:
Yes i will do it when it got more time sorry

AMDrewtaylor, 2025-05-13:
No worries!

zetneteork, 2026-09-03:
Root cause: `install.sh` only matches exact `$ID` values from `/etc/os-release`. Zorin OS has `ID=zorin` with `ID_LIKE="ubuntu debian"`, but the script does not check `$ID_LIKE`.

Fix being implemented: refactoring all `case "$ID" in` blocks in `install.sh` to also check `$ID_LIKE`, so any Debian-derivative (Zorin, Devuan, Pop!_OS, elementary, etc.) is automatically recognized.

Tracking at: https://github.com/opentreecz/jLink/tree/integration/upstream

zetneteork, 2026-09-03:
Fixed in PR #32. The installer now checks `$ID_LIKE` so Zorin OS (`ID_LIKE="ubuntu debian"`) is auto-detected as Debian-based. This can be closed.

Watchdog0x, 2026-09-03:
Resolved in Jabridge v1.0.0-rc.2 and PR #33. The old distribution-specific dependency installer is gone: the release is a portable static executable, and the source installer no longer rejects Linux distribution IDs. Zorin users can unpack the release and run ./jabridge directly. Headset compatibility remains separate from this installation fix.

closed 2026-09-03 by Watchdog0x