Watchdog0x/jabridge #12: Adding OS

voidwatch, 2025-05-21:
Hello!

Awesome repo, thank you so much!
I needed to mod the install.sh to add my dist in the list.
It worked, so if you'd like to update the install.sh - please do, see if you can spot the difference ;)

`        case "$ID" in
            ubuntu|debian)
                dpkg -l | grep -qw "$lib"
                return $?
                ;;
            linuxmint|debian)
                dpkg -l | grep -qw "$lib"
                return $?
                ;;
            fedora|centos|rhel)
                rpm -q "$lib"
                return $?
                ;;
            arch|manjaro)
                pacman -Qs "$lib" >/dev/null
                return $?
                ;;
            *)
                echo "Unsupported distribution: $ID"
                return 1
                ;;
        esac
    }`

All the best
Void

Watchdog0x, 2025-07-02:
Hello Void,

Thanks so much for your message and sorry for the delayed reply! 
You’re absolutely awesome for making that improvement.

Your added linuxmint into the install.sh case block😜

Thank you for fixing it! I’ll go ahead and update the install.sh script with your patch on the next release.

ahoffskov, 2025-10-24:
Maybe it's because I'm not a programmer - but wouldn't changing
`arch|manjaro)`
to
`arch|manjaro|endeavouros)`
be enough to add EndeavourOS support? Or do I need to change other factors too?

zetneteork, 2026-09-03:
Linux Mint support is already merged via PR #26 on the `integration/upstream` branch.

Regarding EndeavourOS (from the comment above): we are refactoring `install.sh` to check `$ID_LIKE` in addition to `$ID`. EndeavourOS has `ID_LIKE=arch`, so it will be automatically detected as Arch-based and use `pacman`. This will also cover other Arch derivatives like Garuda.

Tracking at: https://github.com/opentreecz/jLink/tree/integration/upstream

zetneteork, 2026-09-03:
Linux Mint support merged via PR #26. EndeavourOS and other derivatives are now auto-detected via `$ID_LIKE` fallback from PR #32. This can be closed.

Watchdog0x, 2026-09-03:
Resolved in Jabridge v1.0.0-rc.2 and PR #33. The old distribution case list is no longer used. The release archive contains a portable static executable, while source installation only requires a suitable Go toolchain. Linux Mint, EndeavourOS, and other distributions are therefore not rejected by an installer allowlist.

closed 2026-09-03 by Watchdog0x