Watchdog0x/jabridge #15: Jabra Link 370 not recognized

zendynar, 2025-07-04:
My dongle for Jabra Evolve 75 - Jabra Link 370 -is not recognized:
`ID 0b0e:245e GN Netcom Jabra Link 370`

The program just spins with Looking for Dongle message.
My machine is Ubuntu 24.04.

Please let me know how I can help you debug this, I have 2 pairs of Evolve 75 and I am motivated to get them to work properly under Linux

Watchdog0x, 2025-07-07:
I'm glad to hear that! 
It sounds like you're just missing the appropriate udev rules. 
Once you add them, you should be able to run it without root access.

Here's how you can set it up:

```bash
# Create a udev rule for JLink devices
echo 'ATTRS{idVendor}=="0b0e", MODE="0660", GROUP="users"' | sudo tee /etc/udev/rules.d/99-jabra.rules > /dev/null

# Reload udev rules
sudo udevadm control --reload

# Apply the new rule immediately
sudo udevadm trigger
```

Watchdog0x, 2025-07-07:
Ah, that’s a really good idea,  thanks for the suggestion! 
I might include it in the next update.

I used the users group because I thought it was a standard group across all distros, similar to how root is used. 
That’s why I chose it by default.

As for the id command, it’s a standard GNU utility from Coreutils (https://www.gnu.org/software/coreutils/), like chmod, groups, and others.

zetneteork, 2026-09-03:
Root cause: udev permissions. The current udev rule uses `GROUP="users"`, but on many systems (corporate Ubuntu, Fedora, custom configurations) the user is not in the `users` group. As @zendynar discovered, using the actual user group works.

Fix being implemented: replacing `GROUP="users"` with `TAG+="uaccess"` in the udev rule. This is the modern systemd-logind approach -- it grants device access to whoever is physically logged in, regardless of group membership. No need for `groupadd` or manual group management.

Tracking at: https://github.com/opentreecz/jLink/tree/integration/upstream

zetneteork, 2026-09-03:
Root cause was udev permissions. Fixed in PR #32 by switching the udev rule from `GROUP="users"` to `TAG+="uaccess"` (modern systemd-logind approach that grants access to the physically logged-in user). This can be closed once users confirm the fix.

Watchdog0x, 2026-09-03:
RC5 replaces the old group-based rule with one-command active-user access:

```bash
./jabridge setup
```

Setup installs the rule, reloads and retriggers udev, and verifies access without running the app as root. The original reporter already confirmed the Link 370 works once the permission rule is correct, so I am closing this original permission bug as completed.

RC5: https://github.com/Watchdog0x/jabridge/releases/tag/v1.0.0-rc.5

@zendynar @ottomorac, a normal-user RC5 confirmation would still be welcome in #34. Please open a new model-specific report if setup itself fails.

closed 2026-09-03 by Watchdog0x