Watchdog0x/jabridge #16: Detect / call event when headset turns on

SimFre, 2025-07-13:
I saw there's an idea to create a jLink daemon. Would it also be possible then to detect when a wireless headset turns on / connects?

Specifically, Jabra Evolve 75, using Jabra Link 370.

zetneteork, 2026-09-03:
The Jabra SDK already provides `deviceAttachedFunc` and `deviceRemovedFunc` callbacks, which jLink uses internally to detect when devices connect/disconnect. The missing piece is exposing these events to external tools.

This requires the daemon service architecture described in TODO #5 (background service with IPC). When that is implemented, events could be exposed via a Unix socket or D-Bus, allowing GNOME/KDE applets or scripts to react to headset on/off.

Not fixable in the current architecture, but the groundwork (callbacks, device tracking) is already in the codebase.

Watchdog0x, 2026-09-03:
Implemented in RC5.

The service now publishes `device.attached`, `device.detached`, `device.battery.update`, and `device.pairing.update` events over its private IPC socket. The simple watcher is:

```bash
jabridge ipc watch
```

It keeps the connection alive automatically. The TUI uses the same event service and reconnects after service restarts. I am closing the API/event feature as completed; the specific Evolve 75 + Link 370 hardware behavior can be reported in #34.

closed 2026-09-03 by Watchdog0x