Watchdog0x/jabridge #18: Screen flickering

fansari, 2025-09-25:
I have tried to use  this with "Jabre Evolve2 55" with Fedora 42 Silverblue.

First I was not able to use the install script because it is not aware of os-tree based Linux syystems where there is no dnf but rpm-ostree instead and /usr ist not a writable directory.

I checked what the script is trying to do and put the library file to /usr/local/lib/jabra/libjabra.so.

Then echo "/usr/local/lib/jabra" > /etc/ld.so.conf.d/jabra.conf.

Starting jLink as normal user it tries to find the dongle.

Starting jLink as root shows some menu which is not well readable and is flickering all the time. Running as root it is trying to find the headset.

zetneteork, 2026-09-03:
This has been fixed by PR #28 from @delacor. The fix introduces a `clearLine` function, uses a `boxDrawn` flag to prevent redundant draws, and only clears the screen on size changes.

PR #28 is now merged on the `integration/upstream` branch at https://github.com/opentreecz/jLink/tree/integration/upstream along with CI/CD, packaging, and other improvements.

zetneteork, 2026-09-03:
Fixed by PR #28 from @delacor. The TUI now only clears the screen on size changes using a boxDrawn flag and clearLine function. This can be closed.

Watchdog0x, 2026-09-03:
Resolved in Jabridge v1.0.0-rc.2 and PR #33. The TUI now redraws only after input, a terminal-size change, or an actual state revision; unchanged polling no longer clears the screen. The Link 380 unchanged-state capture rendered once over seven seconds. Real-headset RC2 regression testing continues in #34, so please reopen this issue if RC2 still redraws while its displayed state is unchanged.

closed 2026-09-03 by Watchdog0x